JanLahmann/RasQberry-Two #1: .local directory structure is missing

torstenandrecht, 2024-08-03:
For the error free execution of the command 

wget https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh -O RasQ-init.sh
. ./RasQ-init.sh

the directories .local .local/config and .local/bin are missing.

After the manual creation of the the mentioned directory structure, the shell shell script above runs without a problem.
After the initial config via raspi-config the on-click installation were also possible.

JanLahmann, 2024-08-04:
What error do you get?
The directories should get created in line 69 and 74 of RasQ-init.sh, e.g.
https://github.com/JanLahmann/RasQberry-Two/blob/64e487f966440345a20b3d105be9dacb0dc0dae7/RasQ-init.sh#L69
Maybe something like 
`[ ! -d ~/.local ] && mkdir ~/.local`
is missing or a suitable flag for mkdir. 

But I can’t test it myself right now.

torstenandrecht, 2024-08-06:
Here is the requested error I get:

 wget https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh -O RasQ-init.sh
--2024-08-06 08:37:16--  https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/JanLahmann/RasQberry-Two/main/RasQ-init.sh [following]
--2024-08-06 08:37:16--  https://raw.githubusercontent.com/JanLahmann/RasQberry-Two/main/RasQ-init.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8002::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3718 (3.6K) [text/plain]
Saving to: ‘RasQ-init.sh’

RasQ-init.sh                    100%[=====================================================>]   3.63K  --.-KB/s    in 0s      

2024-08-06 08:37:17 (51.6 MB/s) - ‘RasQ-init.sh’ saved [3718/3718]

rasqberry@rasqberry-two:~ $ . ./RasQ-init.sh

This script (RasQ-init.sh) downloads a modified raspi-config tool,
which can be used to install and configure Qiskit and
several Quantum Computing Demos on the Raspberry Pi
This script is available at https://github.com/JanLahmann/RasQberry-Two/

Before running this script, pls setup the SD card using the
Raspberry Pi Imager (with 'Raspberry Pi OS 64-bit with Desktop'),
then login to the RPi via ssh and run the following commands:

wget https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh -O RasQ-init.sh
. ./RasQ-init.sh
Also use the previous command to update the RasQberry tooling.

The script be run with parameters . ./RasQ-init.sh branch gituser devoption
The -branch parameter is used to specify the branch you want to use
The -gituser parameter is used to specify the github user to use for the rasqberry repository
The -devoption parameter is used to install the development version of the rasqberry repository (production=0, dev=1)
default values are: branch=main gituser=JanLahmann devoption=0
example: . ./RasQ-init.sh main JanLahmann 1

See https://rasqberry.org for an in-depth description
of RasQberry, which combines Raspberry Pi and Qiskit,
IBM's open source Quantum Computing software framework.

using Branch  main and from github-user  JanLahmann with DevOption  0

cloning  RasQberry-Two
using production mode
-bash: git: command not found

replacing raspi-config with the modified version
--2024-08-06 08:37:29--  https://raw.githubusercontent.com/JanLahmann/RasQberry-raspi-config/bookworm/raspi-config
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8002::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 110415 (108K) [text/plain]
Saving to: ‘raspi-config’

raspi-config                    100%[=====================================================>] 107.83K  --.-KB/s    in 0.04s   

2024-08-06 08:37:29 (2.52 MB/s) - ‘raspi-config’ saved [110415/110415]

copy all binaries from the GH repo to ~/.local/bin
mkdir: cannot create directory ‘/home/rasqberry/.local/bin’: No such file or directory

move all config files from the GH repo to ~/.local/config
mkdir: cannot create directory ‘/home/rasqberry/.local/config’: No such file or directory
ln: failed to create symbolic link '/home/rasqberry/.local/bin': No such file or directory

RasQ-init.sh finished
You may now run    sudo raspi-config

=====

ls -la
total 136
drwx------ 2 rasqberry rasqberry   4096 Aug  6 08:37 .
drwxr-xr-x 3 root      root        4096 Aug  3 09:02 ..
-rw-r--r-- 1 rasqberry rasqberry    220 Aug  3 09:02 .bash_logout
-rw-r--r-- 1 rasqberry rasqberry   3523 Aug  3 09:02 .bashrc
-rw-r--r-- 1 rasqberry rasqberry    807 Aug  3 09:02 .profile
-rw-r--r-- 1 rasqberry rasqberry 110415 Aug  6 08:37 raspi-config
-rw-r--r-- 1 rasqberry rasqberry   3718 Aug  6 08:37 RasQ-init.sh
-rw-r--r-- 1 rasqberry rasqberry      0 Aug  6 08:37 .sudo_as_admin_successful
-rw-r--r-- 1 rasqberry rasqberry    215 Aug  6 08:37 .wget-hsts

JanLahmann, 2024-08-06:
I've always tried with the user "pi", not any other user, e.g. rasqberry. 
Yours was an important test. 

I think I've fixed the issue; adding a -p flag to mkdir. Could you please try again?

torstenandrecht, 2024-08-06:
This time it worked:

rasqberry@rasqberry-two:~ $ wget https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh -O RasQ-init.sh
--2024-08-06 10:18:39--  https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/JanLahmann/RasQberry-Two/main/RasQ-init.sh [following]
--2024-08-06 10:18:39--  https://raw.githubusercontent.com/JanLahmann/RasQberry-Two/main/RasQ-init.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8000::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3724 (3.6K) [text/plain]
Saving to: ‘RasQ-init.sh’

RasQ-init.sh        100%[=================>]   3.64K  --.-KB/s    in 0s      

2024-08-06 10:18:40 (28.8 MB/s) - ‘RasQ-init.sh’ saved [3724/3724]

rasqberry@rasqberry-two:~ $ . ./RasQ-init.sh

This script (RasQ-init.sh) downloads a modified raspi-config tool,
which can be used to install and configure Qiskit and
several Quantum Computing Demos on the Raspberry Pi
This script is available at https://github.com/JanLahmann/RasQberry-Two/

Before running this script, pls setup the SD card using the
Raspberry Pi Imager (with 'Raspberry Pi OS 64-bit with Desktop'),
then login to the RPi via ssh and run the following commands:

wget https://github.com/JanLahmann/RasQberry-Two/raw/main/RasQ-init.sh -O RasQ-init.sh
. ./RasQ-init.sh
Also use the previous command to update the RasQberry tooling.

The script be run with parameters . ./RasQ-init.sh branch gituser devoption
The -branch parameter is used to specify the branch you want to use
The -gituser parameter is used to specify the github user to use for the rasqberry repository
The -devoption parameter is used to install the development version of the rasqberry repository (production=0, dev=1)
default values are: branch=main gituser=JanLahmann devoption=0
example: . ./RasQ-init.sh main JanLahmann 1

See https://rasqberry.org for an in-depth description
of RasQberry, which combines Raspberry Pi and Qiskit,
IBM's open source Quantum Computing software framework.

using Branch  main and from github-user  JanLahmann with DevOption  0

cloning  RasQberry-Two
using production mode
-bash: git: command not found

replacing raspi-config with the modified version
--2024-08-06 10:18:48--  https://raw.githubusercontent.com/JanLahmann/RasQberry-raspi-config/bookworm/raspi-config
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8000::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 110415 (108K) [text/plain]
Saving to: ‘raspi-config’

raspi-config        100%[=================>] 107.83K  --.-KB/s    in 0.04s   

2024-08-06 10:18:49 (2.51 MB/s) - ‘raspi-config’ saved [110415/110415]

copy all binaries from the GH repo to ~/.local/bin

move all config files from the GH repo to ~/.local/config

RasQ-init.sh finished
You may now run    sudo raspi-config

closed 2024-10-29 by JanLahmann