JanLahmann/RasQberry-Two #17: Images size need to be belo 2 GByte

torstenandrecht, 2024-10-27:


torstenandrecht, 2024-10-27:
-rw-r--r-- 1 root root 14151581696 Oct 22 17:17 2024-10-22-rasqberry-bookworm--full.img
-rw-r--r-- 1 root root      228988 Oct 22 17:17 2024-10-22-rasqberry-bookworm--full.info
drwxr-xr-x 2 root root        4096 Oct 22 17:06 rootfs
-rw-r--r-- 1 root root  6006243328 Oct 22 17:02 2024-10-22-rasqberry-bookworm-.img
-rw-r--r-- 1 root root      211210 Oct 22 17:02 2024-10-22-rasqberry-bookworm-.info
-rw-r--r-- 1 root root  2864709632 Oct 22 16:55 2024-10-22-rasqberry-bookworm--lite.img
-rw-r--r-- 1 root root       78314 Oct 22 16:55 2024-10-22-rasqberry-bookworm--lite.info

torstenandrecht, 2024-10-27:
@rishi will try to move the Qiskit Installation scripts to an earlier stage.

JanLahmann, 2024-10-29:
removing the additional packages (e.g. Wolfram) in stage 5 and using xz compression instead of zip will probably solve this.
In addition, we should think about the structure of the stages.
I'd suggest we move "our" code to a new stage (stage 6), and try to skip stage 5 in the image build process. Probably, this should be a new issue.

JanLahmann, 2024-10-30:
solved with https://github.com/JanLahmann/RasQberry-pi-gen/pull/16

closed 2024-10-30 by JanLahmann