JanLahmann/RasQberry-Two #33: install & enable docker in RasQberry image

JanLahmann, 2024-11-09:
docker will be used by several demos and should be installed & enabled in the image

torstenandrecht, 2024-11-15:
here is a good description: https://pimylifeup.com/raspberry-pi-docker/ I'll focus on that topic and will do the configuration

JanLahmann, 2024-11-15:
Thank you, Torsten!
In the previous project, the code is here: https://github.com/JanLahmann/RasQberry/blob/0619f91c6ceb43c94e733ff02ef387c1ab9ee72b/config/setup.sh#L244

torstenandrecht, 2024-11-16:
> Thank you, Torsten! In the previous project, the code is here: https://github.com/JanLahmann/RasQberry/blob/0619f91c6ceb43c94e733ff02ef387c1ab9ee72b/config/setup.sh#L244

The installation methoded changed with debian bookworm:

```
# Add Docker's official GPG key:
sudo apt-get update
sudo apt-get install ca-certificates curl
sudo install -m 0755 -d /etc/apt/keyrings
sudo curl -fsSL https://download.docker.com/linux/debian/gpg -o /etc/apt/keyrings/docker.asc
sudo chmod a+r /etc/apt/keyrings/docker.asc

# Add the repository to Apt sources:
echo \
  "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/debian \
  $(. /etc/os-release && echo "$VERSION_CODENAME") stable" | \
  sudo tee /etc/apt/sources.list.d/docker.list > /dev/null
sudo apt-get update

sudo apt-get install docker-ce docker-ce-cli containerd.io docker-buildx-plugin docker-compose-plugin

JanLahmann, 2025-04-25:
solved here: stage-RQB2/04-install-docker

closed 2025-04-25 by JanLahmann