JanLahmann/RasQberry-Two #38: enable LEDs

JanLahmann, 2024-11-09:
Enable LED strips and LED Matrix display for both RPi 4 and RPi 5.

JanLahmann, 2024-11-09:
See https://github.com/JanLahmann/RasQberry/blob/0619f91c6ceb43c94e733ff02ef387c1ab9ee72b/bin/rq_luma_setup.sh in the first RasQberry project

saschaschefenacker, 2024-11-09:
I could have a look into this one

JanLahmann, 2024-11-11:
Thanks for the first code to enable the LEDs!
I think we will use the "default" venv to add such functionality.
As some of the demos will want to use LEDs, we will provide a venv with Qiskit and the basic tooling (incl. LEDs) as part of our basic image.
The name of the standard venv is defined here: https://github.com/JanLahmann/RasQberry-pi-gen/blob/2a03dbea8da55aa75605693aa0523ace922d7c80/config#L20C1-L20C16 `STD_VENV="RQB2"`. Please use the variable, if possible. Otherwise we can adjust it during final integration into the image.

saschaschefenacker, 2024-11-14:
the SPI needs to be enabled in the image:
https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/

manual: sudo raspi-config

gbanchelli, 2024-11-15:
> the SPI needs to be enabled in the image: https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/
> 
> manual: sudo raspi-config

As discussed with @JanLahmann, I will test this and submit a PR for SPI enablement.

saschaschefenacker, 2024-11-20:
Thanks!
Seems https://github.com/JanLahmann/RasQberry-Two/pull/43 would be ready for the initial merge.

JanLahmann, 2025-04-25:
see also #83