JanLahmann/RasQberry-Two #39: create/define environment file to store infos&state

JanLahmann, 2024-11-09:
an environment file in the running image can be used to store infos like number of LEDs, existing SenseHAT, etc
Structure and content needs to be defined, and how to access from shell scripts and python.

The previous RasQberry has implemented this to some extend.

rishimathur14, 2024-11-15:
@JanLahmann can be assigned to me.

JanLahmann, 2025-04-25:
has this been solved with #78 ?

JanLahmann, 2025-10-18:
This has been fully implemented. The environment file system is now in place:

**Implementation:**
- Central configuration: `RQB2-config/rasqberry_environment.env`
- Environment loader: `RQB2-config/env-config.sh`
- All scripts source the configuration via `env-config.sh`

**Features included:**
- System variables (REPO, RQB2_CONFDIR, STD_VENV)
- LED configuration (LED_COUNT, LED_PIN, LED_BRIGHTNESS, etc.)
- Demo repository URLs (GIT_REPO_DEMO_*)
- Demo installation state flags (QUANTUM_LIGHTS_OUT_INSTALLED, etc.)
- Accessible from both shell scripts and Python

The environment file is automatically loaded by all RQB2 scripts and provides centralized configuration management as requested.

Closing as resolved.