JanLahmann/RasQberry-Two #41: add a simple quantum demo using the LEDs

JanLahmann, 2024-11-09:
Add a simple demo using some superposition and entanglement, with display of measurement result on the LED matrix display.
See https://github.com/JanLahmann/RasQberry/blob/0619f91c6ceb43c94e733ff02ef387c1ab9ee72b/demos/bin/RasQ-LED_desktop.sh and https://github.com/JanLahmann/RasQberry/blob/0619f91c6ceb43c94e733ff02ef387c1ab9ee72b/demos/bin/RasQ-LED.py

Luka-D, 2024-12-10:
@JanLahmann Is there an image or a video of this demo running on the RasQberry 1? I am curious what the result looked like on that display?

JanLahmann, 2024-12-11:
not yet. I'll try to make a video next week.

Luka-D, 2025-01-08:
How many pixels will be required for this demo? I tried running the previous RasQ 1 demo and the results I'm getting are a single measurement, 1 or 0.

Luka-D, 2025-03-17:
Hi @JanLahmann, I've been working on refactoring the old project's demo code to work on the RasQberry Two. I think I've got it working, but I'm unsure how we want to visualize the results on the LED. So far I've just got the qubits displayed on the LEDs with blue for a 1 and grey for a 0. Below are a few photos of running option 3 (the get factors simulations) with an 8 qubit circuit size.

[image]

[image]