JanLahmann/RasQberry-Two #45: rasqberry menu entry magically disapear after the initial execution of the raspberry-tie demo

torstenandrecht, 2024-11-16:


JanLahmann, 2024-11-18:
That is probably caused  by an update of the standard raspi-config, which overrides our modifications. We need to confirm and find a way to prevent it. Maybe a check at bootup and re-insert our changes.

cybersecfaith, 2024-11-18:
[raspi-config_w:issue.txt](https://github.com/user-attachments/files/17807124/raspi-config_w.issue.txt)

@torstenandrecht @JanLahmann - Not sure which logs to capture so I've captured the contents of /usr/bin/raspi-config which according to this should be the one. 

```
(RQB2) rasqberry@rasqberry-two:~ $ which raspi-config
/usr/bin/raspi-config
```

If you modify the file extension and remove the first and last lines, you have the full script on my system.

cybersecfaith, 2024-11-18:
Also uploading the journalctl logs @torstenandrecht @JanLahmann 

[journalctl_log_w:issue.txt](https://github.com/user-attachments/files/17807172/journalctl_log_w.issue.txt)

rishimathur14, 2024-11-22:
@JanLahmann Please assign to me

rishimathur14, 2024-11-28:
Not able to recreate the issue Need help

rishimathur14, 2024-12-07:
Fixed. Raised the PR

closed 2025-04-25 by JanLahmann