JanLahmann/RasQberry-Two #61: Permission denied error during pi-gen process in the wallpaper section with a sudo user

torstenandrecht, 2024-12-07:
with a sudo user  I get in the "wallpaper" during the pi-gen process a "permission denied error". Here ar the commands I use 

sudo git clone --branch 30-sw-platform https://github.com/JanLahmann/RasQberry-pi-gen.git
sudo ./build.sh -c config

update-alternatives: using /usr/sbin/iptables-legacy to provide /usr/sbin/iptables (iptables) in auto mode
update-alternatives: using /usr/sbin/ip6tables-legacy to provide /usr/sbin/ip6tables (ip6tables) in auto mode
update-alternatives: using /usr/sbin/iptables-nft to provide /usr/sbin/iptables (iptables) in auto mode
update-alternatives: using /usr/sbin/ip6tables-nft to provide /usr/sbin/ip6tables (ip6tables) in auto mode
update-alternatives: using /usr/sbin/arptables-nft to provide /usr/sbin/arptables (arptables) in auto mode
update-alternatives: using /usr/sbin/ebtables-nft to provide /usr/sbin/ebtables (ebtables) in auto mode
Setting up docker-ce (5:27.3.1-1~debian.12~bookworm) ...
Running in chroot, ignoring request.
Created symlink /etc/systemd/system/multi-user.target.wants/docker.service → /lib/systemd/system/docker.service.
Created symlink /etc/systemd/system/sockets.target.wants/docker.socket → /lib/systemd/system/docker.socket.
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u9) ...
Ending Docker Installation
[10:13:09] End /home/rasqberry/RasQberry-pi-gen/stage5/04-install-docker/00-run-chroot.sh
[10:13:09] End /home/rasqberry/RasQberry-pi-gen/stage5/04-install-docker
[10:13:09] Begin /home/rasqberry/RasQberry-pi-gen/stage5/05-wallpapers
[10:13:09] Begin /home/rasqberry/RasQberry-pi-gen/stage5/05-wallpapers/00-run.sh
mkdir: cannot create directory ‘/home/rasqberry’: Permission denied
[10:13:09] Build failed

torstenandrecht, 2024-12-09:
Error also happens mit UID user 1000 and the folowing commands:

sudo rm -rf RasQberry-pi-gen/
git clone --branch 30-sw-platform https://github.com/JanLahmann/RasQberry-pi-gen.git
cd RasQberry-pi-gen/
sudo ./build.sh

JanLahmann, 2024-12-09:
for me, the exact same commands do work successfully.
We need to identify what the difference is in our environments...

JanLahmann, 2024-12-09:
The usernames and IDs might be important.
I have the following user-IDs (1000 and above) in /etc/passwd :
`rasqberry:x:1000:1000:,,,:/home/rasqberry:/bin/bash`

torstenandrecht, 2024-12-10:
I did the test with the following user on another system and I got the permission denied error again: uid=1000(rasqberry) gid=1000(rasqberry) groups=1000(rasqberry),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi)

update-alternatives: using /usr/sbin/iptables-legacy to provide /usr/sbin/iptables (iptables) in auto mode
update-alternatives: using /usr/sbin/ip6tables-legacy to provide /usr/sbin/ip6tables (ip6tables) in auto mode
update-alternatives: using /usr/sbin/iptables-nft to provide /usr/sbin/iptables (iptables) in auto mode
update-alternatives: using /usr/sbin/ip6tables-nft to provide /usr/sbin/ip6tables (ip6tables) in auto mode
update-alternatives: using /usr/sbin/arptables-nft to provide /usr/sbin/arptables (arptables) in auto mode
update-alternatives: using /usr/sbin/ebtables-nft to provide /usr/sbin/ebtables (ebtables) in auto mode
Setting up docker-ce (5:27.4.0-1~debian.12~bookworm) ...
Running in chroot, ignoring request.
Created symlink /etc/systemd/system/multi-user.target.wants/docker.service → /lib/systemd/system/docker.service.
Created symlink /etc/systemd/system/sockets.target.wants/docker.socket → /lib/systemd/system/docker.socket.
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u9) ...
Ending Docker Installation
[23:01:46] End /home/rasqberry/RasQberry-pi-gen/stage5/04-install-docker/00-run-chroot.sh
[23:01:46] End /home/rasqberry/RasQberry-pi-gen/stage5/04-install-docker
[23:01:46] Begin /home/rasqberry/RasQberry-pi-gen/stage5/05-wallpapers
[23:01:46] Begin /home/rasqberry/RasQberry-pi-gen/stage5/05-wallpapers/00-run.sh
mkdir: cannot create directory ‘/home/rasqberry/RasQberry-pi-gen/work/stage5/rootfs/home/rasqberry/.config/pcmanfm’: Permission denied
[23:01:46] Build failed

JanLahmann, 2024-12-10:
Thank you for testing.  
The error is slightly different. 
Can you share ls -l for the directory /home/rasqberry/RasQberry-pi-gen/work/stage5/rootfs/home/rasqberry/.config/pcmanfm and the parent(s)?

gbanchelli, 2024-12-13:
As Jan suggested, we might be dealing with incorrect permissions on the parent directories.
If that's the case, removing the  work/stage5/rootfs/home/rasqberry/.config subdirectory that's in your build path should allow it to get re-created with the proper ownership and permissions.

@JanLahmann please assign this one to me.

gbanchelli, 2024-12-18:
After testing on a freshly imaged build system, it became clear that the default user's ~/.config subdirectory was being created by the root user at some point in stage 4.
There are three adjacent section of raspi-config that will create this subdirectory if not already present. If raspi-config is run by the root user, the subdirectory will be owned by the root user and group.
The cleanest way to prevent this is to ensure the subdirectory is pre-created via /etc/skel, using the same approach as for .local.

JanLahmann, 2024-12-19:
Thank you for analysing, @gbanchelli !
Could you please check with Rishi and create a PR for this?

JanLahmann, 2025-11-03:
Closing due to age (Dec 2024) and no recent reports. Wallpaper stage working correctly in current builds. Please reopen with reproduction steps if issue persists.