JanLahmann/RasQberry-Two #76: Please consider a Docker installation

oe3gwu, 2025-02-26:
Hello!

After you considered a docker release at version 1, I want to move this open issue the version 2 too. It would make a whole lot easier in order to get this system running.

Thank you

JanLahmann, 2025-02-27:
Thanks a lot for that suggestion, @oe3gwu!
We are still finalising the SW image - and currently focus on a SD image (i.e. a complete SD image with the RasQberry & Qiskit packages included, not a script that installs our code on top a normal Raspberry Pi OS image). 
We also plan to integrate that into the Raspberry Pi Imager, and think this will make install much easier than for the first RasQberry version.

Would that help from your perspective, or do you still see a need for a docker image? We could consider that, but will first focus on the main release (SD image for the RPI Imager).

oe3gwu, 2025-02-27:
@JanLahmann thanks for the response!

1st of all I want to thank you for the project, it is simply great and cant wait to use it again. Got big issues with Rasqberry One! :-)

It generally definately would help, but the think why I insist for docker is simple upgradeability and no python dependency hell. If you make a SD image, that is great, but dependency hell can break loose if you apt update if python packages are installed system wide. Also Qiskit is kinda delicate with what versions of python libraries are installed. With docker you could "freeze" your project in time with a working code, no matter what somebody do to the pi. And if some new version comes out from qiskit or whatever, you can simply roll out a new image.

Assuming you have a docker compose file, you simply can update to the newser version with a 

```bash
docker compose down && docker compose pull && docker compose up -d
```

Building the inital image surly is a pain, I am a platform engineer, I know what I speak of! ;-)

I can definately help as tester, but I wont build an image because I dont want to split or fork your great work.