JanLahmann/RasQberry-Two #83: compare adafruit-circuitpython-neopixel-spi vs rpi_ws281x

JanLahmann, 2025-04-25:
LED support on RPi 5 was a bit challenging.
We are using adafruit-circuitpython-neopixel-spi, but in the meantime the standard rpi_ws281x is available for RPi5 as well (https://github.com/jgarff/rpi_ws281x/wiki/Raspberry-Pi-5-Support). 

Would be good to re-evaluate what LED support to use in RasQberry.

rpi_ws281x might enable Luma (https://github.com/rm-hull/luma.led_matrix/) that provides very convenient use of LED matrix displays (scrolling text, pictures, etc).

Luka-D, 2025-04-25:
I'll do some investigation on this

JanLahmann, 2025-04-25:
see also #38

Luka-D, 2025-04-28:
### Initial results: 

I've followed all the steps listed [here](https://github.com/jgarff/rpi_ws281x/wiki/Raspberry-Pi-5-Support) on my Pi 5. I needed to install cmake using `sudo apt install cmake` because it was not on my os by default. Trying to run the test using `sudo ./test` gives me the following error:
```
ws2811_init failed: Hardware revision is not supported
```

For reference, the hardware specs of your Pi can be found using: `cat /proc/cpuinfo`. Of note are the following:

```
Revision     : b04170
Model        : Raspberry Pi 5 Model B Rev 1.0
```
It seems like my Pi 5 model doesn't support the Pi 5 branch linked above. I will read some issues and forum posts to try to trouble shoot this. I will also try the official rpi-ws281x python release on PyPi as well.

JanLahmann, 2025-05-01:
Thanks for the update, Luka!