JanLahmann/RasQberry-Two #103: error running rasqberry demos

andysc, 2025-05-31:
I don't know how I did it - I can't reproduce it now, but one time it said 

`There was an error running option 0 rasqberry` as a pop-up in raspi-config with an OK option.

I thought it was after I ctrl-C'd out of the rasqberry TIE aer-noise option (second one) as I didn't know my ibm token... and then restarted raspi-config. But that seems to work fine now, so I don't know how it happened.

JanLahmann, 2025-06-01:
Thanks for reporting this unexpected behaviour, @andysc !
Let's keep this issue open, and see if others might experience something similar.

JanLahmann, 2025-11-03:
✅ Resolved

Extensive demo fixes completed in dev-review01 (Oct 2025):
- Fixed QRT, Fractals, RQL, Qoffee-Maker
- Fixed desktop icon execution (stdin buffer flush)
- Fixed root vs user context issues
- Comprehensive testing completed on Pi 4 and Pi 5

All demos now work from both desktop icons and raspi-config.

closed 2025-11-03 by JanLahmann