JanLahmann/RasQberry-Two #105: LED image corruption on IBM demo

andysc, 2025-06-04:
When I launch the IBM LEDs demo from the desktop, the bottom half of the "I" is missing.
The stripey image is fine, and the second time around is fine also. It's just the first time you launch it.

This does not happen when you run it from raspi-config

andysc, 2025-06-04:
I was able to capture an image - this one was even worse - the top of the I was corrupted also

[image]

droot64s, 2025-06-30:
I have the same issue, even with the standard LED test. After running the demo, all the lights turned white, and my Pi4B crashed (I still cannot turn it on).

andysc, 2025-06-30:
Are you using a decent power supply for the Pi ? When they are "full on" the LEDs pull quite a bit of juice. And when they occasionally go "all white" that's the worst situation (all three colours full on, for every pixel). Is this with a Pi 4, or a Pi 5, out of interest?

andysc, 2025-06-30:
Power supply should be fine.
Testing on the Pi 4 was limited for this version... I'll give it a spin on my rig and see how it looks.

JanLahmann, 2025-11-03:
✅ Fixed

Commit 0ea2b36: "fix(ibm-demo): use environment-configured LED layout mapping"

IBM demo now uses layout-aware mapping from rq_led_utils.