JanLahmann/RasQberry-Two #106: LEDs have R and G swapped round

andysc, 2025-06-04:
I noticed it in the TIE demo, as that has the pi HAT simulator on screen to show the intended colours.

But it looks like in Quantum lights out and Quantum TIE, that 
`neopixel.RGB`
in the code should be changed to 
`neopixel.GRB`

in `RasQberry-Two/Quantum-Lights-Out/lights_out.py` (and `turn_off_leds.py`)
in `RasQberry-Two/demos/quantum-lights-out/lights_out.py` (and `turn_off_leds.py`)

(why are there two versions of this code??)

And in `... demos/TIE/QuantumRaspberryTie.qk1.py`

I notice the files in `..demos/Quantum-Lights_Out/` are owned by root.root - this is not good for a running system.

KPRoche, 2025-08-06:
Hi Luka -- that constant tells the Neo driver about the pixel *hardware*, so it knows _how_ to do the encoding going out to the pixels; the data fed _into_ the assorted functions is always RGB.

Remember that Raspberry Tie is set up to work with various displays, not just Rasqberry, so I included an option to specify which kind of pixels were in the hardware. I have both RGB and GRB arrays in my hardware collection.

Luka-D, 2025-08-13:
@JanLahmann or @rishimathur14